eloquentarduino/EloquentEsp32cam #12: Following your tutorial and hit a snag

iplayfast, 2024-01-15:
Following this tutorial https://eloquentarduino.com/posts/esp32-cam-object-detection
I installed the eloquent library and found the example code for Collect_Images_for_EdgeImpulse 
Tried compiling and found ./xclk.h was not found.

iplayfast, 2024-01-15:
Found the problem, You are probably developing on windows were files aren't case sensitive.
you need to rename:

XCLK.h to xclk.h
Brownout.h to brownout.h
Resolution.h to resolution.h
Sensor.h to sensor.h
Pinout.h to pinout.h

in image_collection.h, mjpeg.h, face_stream.h, decoder_gray, and fomo_stream you need to change include ../camera/Camera.h to ../camera/camera.h

This allows the compile to complete.

eloquentarduino, 2024-01-26:
Fixed in 2.7 (hopefully).

closed 2024-01-26 by eloquentarduino