eloquentarduino/EloquentEsp32cam #15: Camera does not initialize

reymalahay, 2024-01-26:
hi there,

**Is your feature request related to a problem? Please describe.**
i successfully compile and upload the example for object detection. but the camera does not initialize, therefore my camera cannot perform object detection. instead i get this error in the logs:
```
22:37:37.414 -> Cannot init camera
22:37:37.447 -> E (5830) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
22:37:37.447 -> [  5833][E][exception.h:61] set(): [Camera] Cannot init camera
22:37:37.447 -> Cannot init camera
```

**Describe the solution you'd like**
the camera should initialize during setup().

**Describe alternatives you've considered**
n/a

**Additional context**
device: ESP32-WROVER
OS: Linux
IDE: Arduino Studio

please advise.

thanks in advance,
rey malahay

eloquentarduino, 2024-01-26:
Looks like the camera cannot be found. Please post a link to the board you're using. Is the default CameraWebServer example working?

reymalahay, 2024-01-26:
hi @eloquentarduino ,

thank you for the response.

up until `[arduino-esp32](https://github.com/espressif/arduino-esp32)` version `2.0.0` the examples work fine, the `CameraWebServer` example works fine. then from `[arduino-esp32](https://github.com/espressif/arduino-esp32)` version `2.0.1` the camera is not detected.

the board details for `ESP32-Wrover` can be found [here](https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf)

thanks in advance,
rey malahay

> Looks like the camera cannot be found. Please post a link to the board you're using. Is the default CameraWebServer example working?

reymalahay, 2024-01-26:
hi @eloquentarduino,

i raised this `ESP32 WRover` issue with `Edge Impulse`, and they said that they do not currently support the `ESP32-WRover` board. but they did give me an alternative solution. details [here](https://forum.edgeimpulse.com/t/edge-supported-device-that-is-compatible-with-esp32-wrover/9567/2?u=reymalahay)

kind regards,
rey malahay

eloquentarduino, 2024-01-26:
The error you get is not related to Edge Impulse at all. It fails at initializing the camera. You linked to the ESP32 chip you're using, but since you are having troubles with the camera, I guess you're using a board with a camera. Can you link to that board? I develop on espressif core 2.0.14.