eloquentarduino/EloquentEsp32cam #22: collectionServer question on resolution

Tomas1337, 2024-03-07:
Great example on using the camera itself to conveniently get training data.

On your example, you mention that "In fact, you can well capture images at the resolution you prefer (up to the RAM limits, of course) and still have space to run FOMO model.". 

When doing capture with a modified `camera.resolution.face()` to say `.vga()`, the capture interface only seems to be saving the a portion of the streamed image(s). Seems to crop on the top right corner to what I assume is the default resolution of `.face()`

I couldn't find methods to modify this, so I thought i'd ask you first.

medamarko, 2024-04-19:
It seems there is hardocded res in html file:  #jpeg {width: 240px; height: 240px;}, also would be good ot have configurable timer of image taking. currently its hardcoded in html setInterval(() => this.collect(), 250).

eloquentarduino, 2024-06-18:
Will add the possibility to configure.

closed 2024-12-23 by eloquentarduino