eloquentarduino/EloquentEsp32cam #24: Error DRAM overflow

marcovilla118, 2024-03-20:
Dear All,

I'm trying to use AITHinker esp32-cam but it looks like it's not working because of missing memory. Here is the error, do I have any chance to make it work?

/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /private/var/folders/c8/12c3wry92k15j8yj7tgz7rh00000gn/T/arduino/sketches/6F41C98F27E8199FE5E7451F48149081/PersonDetectionExample.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 1192 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

iHaiduk, 2024-04-16:
I have the same error

akvik, 2024-04-18:
Same error here - running the Freenove Wrover with builtin camera.

eloquentarduino, 2024-06-18:
Person detection, face detection, maybe motion detection (can't recall) only work on S3 chips. Also, person detection has been deprecated, you are using an old version of the library.

akvik, 2024-06-18:
Thanks, I figured S3 was needed in the end. I got motion detection to work on non-S3 chip - and I'm following your development of the library - looks great :)