eloquentarduino/EloquentEsp32cam #28: fatal error: dl_image.hpp: No such file or directory

htarahi, 2024-04-24:
Dear friends
By choosing "Motion detection" program on Arduino 1.8.9, I face an error :
`C:\Users\--\Documents\Arduino\libraries\EloquentEsp32cam\src/eloquent_esp32cam/motion/detection.h:4:24: fatal error: dl_image.hpp: No such file or directory`
so what's the problem ?!

jksemple, 2024-05-05:
I think you need to update to Arduino 2.0.x  The dl_image.hpp include file is part of espressif/esp-dl, which may not have been in earlier Arduino releases. This is not really an issue with EloquentESP32cam.

eloquentarduino, 2024-06-18:
You don't actually need Arduino 2.0.x. You need **ESP32 core 2.0.x** (I use Arduino IDE 1.8.x)

closed 2024-06-18 by eloquentarduino

MasterPlexus, 2025-02-01:
I have the same compile error, but I could not find the Library to install. Is the name may slight different? For what I have to search? I use Arduino ide 2.3.4

Thanks for every help!

eloquentarduino, 2025-02-02:
Which ESP32 core version do you have installed?

MasterPlexus, 2025-02-02:
> Which ESP32 core version do you have installed?

I'm really sorry, but I did not find "ESP32 core" in the library manager, but also not as an available option when I try to search for it. Where could I check that (Really sorry, looks like I'm more noob than I expected :-/)

MasterPlexus, 2025-02-02:
ok, sorry, now I see this means Board, and not librarys. I'm switching now from 3.x.y to 2.x.y version and will retry.

MasterPlexus, 2025-02-02:
Yes, thats it. thanks.