eloquentarduino/EloquentEsp32cam #38: object detection and save image to micro SD-ESP32CAM

flix-ui, 2024-07-17:
hello, I have used code for object detection using the FOMO (edgeimpulse) method. Can you help me to combine it with the code to periodically save photos from the esp32cam to the micro SD every 10 minutes?

Thank You
I'm waiting for the good news

flix-ui, 2024-07-19:
okay thankyou for your feedback. i just tried from your reference in https://eloquentarduino.com/posts/esp32-cam-save-pictures-to-sd-card. but I got error (reset (TG1WDT_SYS_RESET) ). i use AI thinker ESP32-CAM.

Thankyou

eloquentarduino, 2024-07-19:
Did you set the correct pins for the SD? By the way, you can use any code
from Google to write to SD card if mine is throwing a error

closed 2024-08-10 by eloquentarduino