eloquentarduino/EloquentEsp32cam #40: JPG decode error when using motion detection and wifi

hbannw, 2024-08-10:
Hi,

I just tested the motion detection example, works fine as is. The program runs for hours (made 200 000 loops so far)
I then added wifi to send the detected capture to a server.
When I connect the wifi, the detection fails with a JPG decode error after 500 to 1500 loops and then the ESP needs to be restarted

Do you have an idea of what's going wrong ? domr tips ?

eloquentarduino, 2024-08-10:
I don't know what could cause the problem. May be some memory leak, maybe? Can you share the sketch you're using?

hbannw, 2024-08-10:
Here is the sketch, the settings.h contains the Wifi credentials, without the wifi lines, it works fine
```
/**
 * Motion detection
 * Detect when the frame changes by a reasonable amount
 *
 * BE SURE TO SET "TOOLS > CORE DEBUG LEVEL = DEBUG"
 * to turn on debug messages
 */
// Include WiFi and MQTT PubSub
#include <WiFi.h>
#include <PubSubClient.h>

#include <settings.h>

#include <eloquent_esp32cam.h>
#include <eloquent_esp32cam/motion/detection.h>

WiFiClient espClient;
PubSubClient client(espClient);

using eloq::camera;
using eloq::motion::detection;
using eloq::wifi;
unsigned long motionTestCount = 0;

/**
 *
 */
void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println("___MOTION DETECTION___");

  // camera settings
  // replace with your own model!
  camera.pinout.aithinker();
  camera.brownout.disable();
  camera.resolution.vga();
  camera.quality.high();

  // configure motion detection
  // the higher the stride, the faster the detection
  // the higher the stride, the lesser granularity
  detection.stride(1);
  // the higher the threshold, the lesser sensitivity
  // (at pixel level)
  detection.threshold(5);
  // the higher the threshold, the lesser sensitivity
  // (at image level, from 0 to 1)
  detection.ratio(0.2);
  // optionally, you can enable rate limiting (aka debounce)
  // motion won't trigger more often than the specified frequency
  detection.rate.atMostOnceEvery(5).seconds();

  // init camera
  while (!camera.begin().isOk())
    Serial.println(camera.exception.toString());

  Serial.println("Camera OK");
  Serial.println("Awaiting for motion...");
    // connect to WiFi
  while (!wifi.connect().isOk())
      Serial.println(wifi.exception.toString());
}

/ *
 */
void loop() {
  // capture picture
  if (!camera.capture().isOk()) {
    Serial.println(camera.exception.toString());
    return;
  }

  // run motion detection
  motionTestCount++;
  if (!detection.run().isOk()) {

    Serial.print("Detect motion error, MotionTestCount =  ");
    Serial.print(motionTestCount);
    Serial.print(" exception = ");
    Serial.println(detection.exception.toString());
    Serial.println("[APP] Free memory: " + String(esp_get_free_heap_size()) + " bytes");
    return;
  }

  // on motion, perform action
  if (detection.triggered()) {
    Serial.println("Motion detected!");
 //   send_photo();
  }
  camera.free();
}
```

eloquentarduino, 2024-08-16:
It is a hardware issue. I tried to run your sketch on my AiThinker ESP32 camera and:

 - sometimes it works
 - sometimes not

Same exact sketch, just resetting the board I get different outcomes. My suggestion? Stop using this awful board and upgrade to a ESP32S3 board: much less troubles.

If you **have to** keep this board, your best bet is to add 

```cpp
delay(10);
yield();
```

at the beginning of `loop()`. I found these lines improve reliability a bit,. but they don't solve the problem. Here's a log from my Serial Monitor: 

```
>>>>>>>> RUN 1: OK <<<<<<<<<<<<

[   473][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
___MOTION DETECTION___
Camera OK
Awaiting for motion...

Connecting to SSID
.........[  6578][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
....
WiFi connected
IP address: 
192.168.110.215
[  8366][I][mutex.h:44] threadsafe(): [Mutex] Creating mutex Camera
Detect motion error, MotionTestCount =  1 exception = Still 4 frames to skip...
[APP] Free memory: 4286171 bytes
Detect motion error, MotionTestCount =  2 exception = Still 3 frames to skip...
[APP] Free memory: 4286171 bytes
Detect motion error, MotionTestCount =  3 exception = Still 2 frames to skip...
[APP] Free memory: 4286171 bytes
Detect motion error, MotionTestCount =  4 exception = Still 1 frames to skip...
[APP] Free memory: 4286171 bytes
Detect motion error, MotionTestCount =  5 exception = Still 0 frames to skip...
[APP] Free memory: 4286171 bytes
[  8768][I][rgb_565.h:101] convert(): [Camera] Allocating 9600 bytes for 80x60 RGB565 image
Detect motion error, MotionTestCount =  6 exception = First frame, can't detect motion
[APP] Free memory: 4266943 bytes
MotionTestCount = 100
MotionTestCount = 200
MotionTestCount = 300
MotionTestCount = 400
MotionTestCount = 500
MotionTestCount = 600
MotionTestCount = 700
MotionTestCount = 800
MotionTestCount = 900
MotionTestCount = 1000
MotionTestCount = 1100
MotionTestCount = 1200
MotionTestCount = 1300
MotionTestCount = 1400
MotionTestCount = 1500
MotionTestCount = 1600
MotionTestCount = 1700
MotionTestCount = 1800
MotionTestCount = 1900
MotionTestCount = 2000
MotionTestCount = 2100
Motion detected!
Detect motion error, MotionTestCount =  2149 exception = Rate limit exceeded. Retry in 4880ms
[APP] Free memory: 4266943 bytes
Detect motion error, MotionTestCount =  2150 exception = Rate limit exceeded. Retry in 4761ms
[APP] Free memory: 4266943 bytes
[... 133 lines cut ...]
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  277 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  278 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  279 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  280 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  281 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  282 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  283 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  284 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  285 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  286 exception = 
[APP] Free memory: 4266651 bytes
Detect motion error, MotionTestCount =  287 exception = 
```

Run 1 and 2 were OK, run 3 went in error. It looks like it is not really a JPEG decode error, but a capture problem.

hbannw, 2024-08-16:
Thanks a lot for testing, I will order a ESP32S3 soon.
For the moment, I set up a workaround using MotionEye on HA with a basic RSP camera setting.

hbannw, 2024-08-23:
Hi Simone, 
I just got a ESP32S3 board but I cannot figure how to connect to the PC, I get an error 43 on the USB controller (DEVICE_DESCRIPTOR_FAILURE:00FF2000)
the board is the following : https://www.amazon.fr/dp/B0CW64341G?ref=ppx_yo2ov_dt_b_fed_asin_title
I certainly miss something...

jksemple, 2024-08-23:
To program it you need to press and hold Boot, then press and release EN/Reset.
You also need to ensure your USB port can supply enough power. You may need a powered hub.
If you are still getting errors you may have a defective board. It's always a good idea to buy two boards so you can do a swap test.

jksemple, 2024-08-23:
And try a different usb cable

hbannw, 2024-08-23:
Already tried 3 different USB cables (they are working with another ESP32 (not S3) board
tried other ports
tried also with another PC, same error

eloquentarduino, 2024-08-23:
First things first. @hbannw are you able to flash any sketch on the board? E.g. Blink

hbannw, 2024-08-23:
No, impossible, the serial port does not show up on the PC
when I plug in the cable, in the PC, I get the error 43 in device manager (there is a new line showing the error)
there is no change when I press EN and boot
how can I check if the PC delivers enough power ?

jksemple, 2024-08-23:
If you've tried the device on a different usb port and it doesn't work there either it is likely to be a defective device rather than lack of power from the usb port.

eloquentarduino, 2024-08-26:
I'm sorry @hbannw , I cannot help with hardware errors.

hbannw, 2024-08-26:
Thank you @eloquentarduino , I just wondered if I missed something, it seems to be a hardware proble so I will return the board

eloquentarduino, 2024-08-30:
do you have `camera.pinout.wroom_s3()`? Also, if not working, update to latest version of EloquentEsp32Cam (2.7.14) and replace `camera.pinout.xxx` with `camera.pinout.autodetect()`.

hbannw, 2024-08-30:
I just tested it, the board reboots aveter the pinout testing

============ Before Setup End ============
19:00:46.123 -> ___MOTION DETECTION___
19:00:46.123 -> [  1650][I][pinout.h:397] test(): [Pinout] Testing pinout for model aithinker
19:00:46.123 -> [  1659][I][pinout.h:397] test(): [Pinout] Testing pinout for model wroom_s3
19:00:46.123 -> [  1666][I][pinout.h:397] test(): [Pinout] Testing pinout for model xiao
19:00:46.156 -> [  1673][I][pinout.h:397] test(): [Pinout] Testing pinout for model M5
19:00:46.156 -> [  1680][I][pinout.h:397] test(): [Pinout] Testing pinout for model M5 wide
19:00:46.156 -> [  1687][I][pinout.h:397] test(): [Pinout] Testing pinout for model M5 timer
19:00:46.156 -> [  1694][I][pinout.h:397] test(): [Pinout] Testing pinout for model ESP-EYE
19:00:46.156 -> [  1701][I][pinout.h:397] test(): [Pinout] Testing pinout for model ESP-EYE S3
19:00:46.192 -> [  1709][I][pinout.h:397] test(): [Pinout] Testing pinout for model Wrover
19:00:46.192 -> [  1716][I][pinout.h:397] test(): [Pinout] Testing pinout for model TTGO LCD
19:00:46.192 -> [  1723][I][pinout.h:397] test(): [Pinout] Testing pinout for model TTGO PIR
19:00:46.192 -> [  1730][I][pinout.h:397] test(): [Pinout] Testing pinout for model T-CAMERA S3
19:00:46.192 -> [  1737][I][pinout.h:397] test(): [Pinout] Testing pinout for model SIMCAM
19:00:46.225 -> ESP-ROM:esp32s3-20210327
19:00:46.225 -> Build:Mar 27 2021
19:00:46.225 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
19:00:46.225 -> Saved PC:0x40376f58
19:00:46.225 -> SPIWP:0xee
19:00:46.225 -> mode:DIO, clock div:1
19:00:46.225 -> load:0x3fce3818,len:0x109c

eloquentarduino, 2024-08-30:
looks like no pinout matched. are you sure the camera is connected properly? Also, try to run the CameraWebServer example from the ESP32 official repo with the WROOM_S3 pin definitions. Does it work?

hbannw, 2024-08-30:
I just tested that, it works using CAMERA_MODEL_ESP32S3_EYE which seems to be the same as wroom_s3 

When I try the example Motion_Detection (from the last version) with wroom_s3, I get a constant reboot

eloquentarduino, 2024-08-30:
Did you enabled Opi PSRAM?

hbannw, 2024-08-30:
Yes, I did

hbannw, 2024-08-31:
I tested another sketch : https://github.com/mihvoi/ESP32S3CamTimeLapse

the results are below, the PSRAM is not detected ? I will contact the vendor

09:14:52.944 -> Build:Mar 27 2021
09:14:52.944 -> rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
09:14:52.944 -> SPIWP:0xee
09:14:52.944 -> mode:DIO, clock div:1
09:14:52.944 -> load:0x3fce3818,len:0x109c
09:14:52.944 -> load:0x403c9700,len:0x4
09:14:52.944 -> load:0x403c9704,len:0xb50
09:14:52.944 -> load:0x403cc700,len:0x2fd0
09:14:52.944 -> entry 0x403c98ac
09:14:53.209 -> 
09:14:53.209 -> 
09:14:53.209 -> Starting ESP32 timelapse
09:14:53.209 -> Remapping SD_MMC card pins to sd_clk=39 sd_cmd=38 sd_data=40
09:14:53.209 -> Pin change success!
09:14:53.241 -> Card Mount SUCCESS
09:14:53.241 -> Card Mount OK
09:14:53.241 -> SD Card Type: SDHC
09:14:53.241 -> SD Card Size: 15113MB
09:14:53.241 -> Total space: 15098MB
09:14:53.241 -> Used space: 0MB
09:14:53.241 -> After initFileSystem SD
09:14:53.241 -> 
09:14:53.241 -> SD card used= 0% ; freeMB=15098
09:14:53.241 -> 
09:14:53.241 -> Warning, PSRAM not detected. If your board has PSRAM, activate it from Tools/PSRAM ('OPI PRAM' option worked for me) !
09:14:53.959 -> E (809) cam_hal: cam_dma_config(301): frame buffer malloc failed
09:14:53.959 -> E (809) cam_hal: cam_config(389): cam_dma_config failed
09:14:53.959 -> E (810) gdma: gdma_disconnect(238): no peripheral is connected to the channel
09:14:53.999 -> E (816) camera: Camera config failed with error 0xffffffff
09:14:53.999 -> Camera init failed with error 0xffffffff

jksemple, 2024-08-31:
You need to enable PS ram in the Arduino IDE
https://www.upesy.com/blogs/tutorials/get-more-ram-on-esp32-with-psram#:~:text=On%20the%20Arduino%20IDE%2C%20to,that%20the%20PSRAM%20is%20activated.

hbannw, 2024-08-31:
Ok, you were right, missed this for the example.
Checked everything with the Motion_Detection example but no luck, constant reboot

eloquentarduino, 2024-09-02:
Please share:

 1. a sketch that uses camera that is working for you
 2. your IDE Tools configuration

hbannw, 2024-09-02:
I will do this next week, actually on vacation

Please share:

hbannw, 2024-09-06:
Hi,

The sketch that works is the following : 
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer
the only changes from the example are:
 #define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM
and set debug level to none otherwise the sketch is too large

I can stream images up to QSXGA size

the IDE config : 
[image]

eloquentarduino, 2024-09-09:
Are you able to run the Take_Picture example using `camera.pinout.eye_s3()` and the same configs?

hbannw, 2024-09-10:
Tested the Take_picture example, reboot loop

Here are the debug outputs
09:29:49.143 -> �ESP-ROM:esp32s3-20210327
09:29:49.143 -> Build:Mar 27 2021
09:29:49.143 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
09:29:49.180 -> Saved PC:0x40376ea4
09:29:49.180 -> SPIWP:0xee
09:29:49.180 -> mode:DIO, clock div:1
09:29:49.180 -> load:0x3fce3818,len:0x109c
09:29:49.180 -> load:0x403c9700,len:0x4
09:29:49.180 -> load:0x403c9704,len:0xb50
09:29:49.180 -> load:0x403cc700,len:0x2fd0
09:29:49.180 -> entry 0x403c98ac
09:29:49.269 -> [   108][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled
09:29:49.269 -> =========== Before Setup Start ===========
09:29:49.269 -> Chip Info:
09:29:49.269 -> ------------------------------------------
09:29:49.303 ->   Model             : ESP32-S3
09:29:49.303 ->   Package           : 0
09:29:49.303 ->   Revision          : 2
09:29:49.303 ->   Cores             : 2
09:29:49.303 ->   CPU Frequency     : 240 MHz
09:29:49.303 ->   XTAL Frequency    : 40 MHz
09:29:49.303 ->   Features Bitfield : 0x00000012
09:29:49.303 ->   Embedded Flash    : No
09:29:49.303 ->   Embedded PSRAM    : No
09:29:49.303 ->   2.4GHz WiFi       : Yes
09:29:49.303 ->   Classic BT        : No
09:29:49.303 ->   BT Low Energy     : Yes
09:29:49.303 ->   IEEE 802.15.4     : No
09:29:49.336 -> ------------------------------------------
09:29:49.336 -> INTERNAL Memory Info:
09:29:49.336 -> ------------------------------------------
09:29:49.336 ->   Total Size        :   393732 B ( 384.5 KB)
09:29:49.336 ->   Free Bytes        :   365816 B ( 357.2 KB)
09:29:49.336 ->   Allocated Bytes   :    23076 B (  22.5 KB)
09:29:49.336 ->   Minimum Free Bytes:   360768 B ( 352.3 KB)
09:29:49.336 ->   Largest Free Block:   327668 B ( 320.0 KB)
09:29:49.336 -> ------------------------------------------
09:29:49.370 -> SPIRAM Memory Info:
09:29:49.370 -> ------------------------------------------
09:29:49.370 ->   Total Size        :  8388608 B (8192.0 KB)
09:29:49.370 ->   Free Bytes        :  8385672 B (8189.1 KB)
09:29:49.370 ->   Allocated Bytes   :      576 B (   0.6 KB)
09:29:49.370 ->   Minimum Free Bytes:  8385672 B (8189.1 KB)
09:29:49.370 ->   Largest Free Block:  8257524 B (8064.0 KB)
09:29:49.370 ->   Bus Mode          : OPI
09:29:49.370 -> ------------------------------------------
09:29:49.370 -> Flash Info:
09:29:49.404 -> ------------------------------------------
09:29:49.404 ->   Chip Size         : 16777216 B (16 MB)
09:29:49.404 ->   Block Size        :    65536 B (  64.0 KB)
09:29:49.404 ->   Sector Size       :     4096 B (   4.0 KB)
09:29:49.404 ->   Page Size         :      256 B (   0.2 KB)
09:29:49.404 ->   Bus Speed         : 80 MHz
09:29:49.404 ->   Bus Mode          : QIO
09:29:49.404 -> ------------------------------------------
09:29:49.404 -> Partitions Info:
09:29:49.404 -> ------------------------------------------
09:29:49.437 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
09:29:49.437 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
09:29:49.437 ->                app0 : addr: 0x00010000, size:  1920.0 KB, type:  APP, subtype: OTA_0
09:29:49.437 ->                app1 : addr: 0x001F0000, size:  1920.0 KB, type:  APP, subtype: OTA_1
09:29:49.437 ->              spiffs : addr: 0x003D0000, size:   128.0 KB, type: DATA, subtype: SPIFFS
09:29:49.471 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
09:29:49.471 -> ------------------------------------------
09:29:49.471 -> Software Info:
09:29:49.471 -> ------------------------------------------
09:29:49.471 ->   Compile Date/Time : Sep 10 2024 09:28:06
09:29:49.471 ->   Compile Host OS   : windows
09:29:49.471 ->   ESP-IDF Version   : v5.1.4-586-gb6b4727c58-dirty
09:29:49.471 ->   Arduino Version   : 3.0.4
09:29:49.505 -> ------------------------------------------
09:29:49.505 -> Board Info:
09:29:49.505 -> ------------------------------------------
09:29:49.505 ->   Arduino Board     : ESP32S3_DEV
09:29:49.505 ->   Arduino Variant   : esp32s3
09:29:49.505 ->   Arduino FQBN      : esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=16M,PartitionScheme=min_spiffs,DebugLevel=debug,PSRAM=opi,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default
09:29:49.548 -> ============ Before Setup End ============

eloquentarduino, 2024-09-10:
This may be the problem: `Arduino Version : 3.0.4`
I tested the library on ESP32 core version 2.x, If I'm correct, other users reported that it didn't work on 3.x (don't know what the problem is).
Go to Board Manager and downgrade to 2.x.

hbannw, 2024-09-10:
Ok, I downgraded to 2.0.17 and it works now for the first tests : Take_Picture works and a basic motion test also, 
Il will make more tests with Wifi and MQTT

Thanks a lot for your time.

hbannw, 2024-09-11:
Tested all night with Wifi and MQTT more than 500000 detection loops, works fine.