eloquentarduino/EloquentEsp32cam #42: Help to use 16Mb flash

Sig2018, 2024-10-15:
First of all thank you for a great Job.It took me a long time to find a library that doesn't use web server.
I'm using a clone of S3 eye and it has 16Mb flash memory.
It works fine. A recognition takes about 901ms.
Is it possible to modify the program to make it use the full memory (16Mb) and not only standard 3Mb App and 1Mb Spiffs?

eloquentarduino, 2024-10-16:
```
Tools > Flash size > 16 MB
Tools > Partition scheme > Choose one of the 2 with 16 MB
```

from the Arduino IDE menu.

Sig2018, 2024-10-16:
Thank you for your quick answer.
Unfortunately none of the two partitions included in Arduino have Spiffs so when I try to use those on Face recognition sketch  it is not possible.

eloquentarduino, 2024-10-16:
Oh, I see. Do they support LittlelFS?

Sig2018, 2024-10-16:
Found the answer my self.
There are other partitions for different custom boards.
There is one with 16Mb total : two 6Mb app and spiffs.
You can copy it on your sketch or customize Arduino IDE to make it available trough Tools>partitions.

closed 2024-10-18 by eloquentarduino