eloquentarduino/EloquentEsp32cam #44: Nice to have feature: Exif data in saved files

morgan-wild, 2024-10-21:
Firstly, thank you so much for your library!

I'm using it to take and store pictures to an SD card, with the date & time as name.
But I'm not able to change the "create date" property in the exif data of these pictures which is wrong.

Exif data are very usefull for many reason (order in an album like the Google Drive for example).

Maybe a new and optional parameter could be added to use sdmmc.save(filePath, *metaData*) ?

Thank you again :)

eloquentarduino, 2024-10-21:
Looks like a reasonable feature. I need to look up the EXIF format. Do you already have a reference implementation?

morgan-wild, 2024-10-21:
Thank you for your fast reply.

In my idea the parameter *metaData* or *exif* could be a structure containing the properties listed here:
https://en.wikipedia.org/wiki/Exif

I looked for an implementation, and maybe there is something helping you here:
https://github.com/dimhoff/ESP32-CAM_Interval/blob/master/exif_defines.h

eloquentarduino, 2024-11-02:
Oh, Sorry, forgot to reply. This is on my todo list, but I'm not actively working on this. Don't expect to see it implemented before a few weeks.

morgan-wild, 2024-11-02:
No problem.

I will use a trick to let you the time to implement it gracefully ;)

Thank you:)