eloquentarduino/EloquentEsp32cam #47: human_face_detect_msr01.hpp: No such file or directory

BimoSora2, 2024-12-21:
there is no this library, you are missing that library in ai thinker esp32cam
```
In file included from /home/gonit/Dokumen/Project file/94/file/file.ino:10:
/home/gonit/Arduino/libraries/EloquentEsp32cam/src/eloquent_esp32cam/face/detection.h:4:10: fatal error: human_face_detect_msr01.hpp: No such file or directory
    4 | #include <human_face_detect_msr01.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
```

eloquentarduino, 2024-12-23:
Which ESP32 Core version do you have installed? This library requires 2.0.x.

BimoSora2, 2024-12-24:
> Which ESP32 Core version do you have installed? This library requires 2.0.x.

I have installed the new library but what I am confused about is that only the ESP32S3 CAM LCD board can find it and the other ESP32S3 don't.

eloquentarduino, 2024-12-24:
To report the ESP32 Core version:

 1. go to `Tools > Board > Board Manager`
 2. search for `ESP32`
 3. look for the one from `Espressif Systems`. It will have the version number listed, something like `2.0.x` or `3.0.x`

Which version do you have?

apellizcos, 2025-05-11:
> To report the ESP32 Core version:
> 
>     1. go to `Tools > Board > Board Manager`
> 
>     2. search for `ESP32`
> 
>     3. look for the one from `Espressif Systems`. It will have the version number listed, something like `2.0.x` or `3.0.x`
> 
> 
> Which version do you have?

Muchas gracias resolvió mi problema con esp32 lilygo v1.7 cam