eloquentarduino/EloquentEsp32cam #48: False postive Face recognition

Sig2018, 2025-01-13:
I'm using the example Face_recognition of the library EloquentEsp32cam
I have no trouble on compilation and it starts fine on ESP32S3 CAM but it is very inaccurate with Face recognition.
I get several face recognized that are not right. 
I'm using my face stored and when I point to any face from a random pic  in my PC monitor most of the time I got my face recognized.
Is there any improvement I can do?
Ive used the ESP32 Camera web server and it is working better, I mean no false positives but I don't want the web server running on ESP32 and your library is a very good option if not where for the false recognition.
Any help will be appreciated.

eloquentarduino, 2025-01-13:
That's strange, since my library is just a wrapper around the ESP-IDF library. I'll take time to review this, but I'll back to work next week, so you'll have to be patient.

Sig2018, 2025-01-13:
OK Thank you for your fast answer.
Let me know if you need any input from my side.
I'm using and ESP32S3 N16R8 and PT 6.25MB app 3.43MB Spiffs.
Since it is my first time with this board I don't know if it is correct, however I don't think it is an issue because there are no errors , just the false recognition.