eloquentarduino/EloquentEsp32cam #49: Problem of init Cam for a Xiao

MasterPlexus, 2025-02-16:
Hi,

I try to reuse the same sketch wich is working for my AIthinker now on a Xiao-ESP32S3 sense.

unfortunalty, even that I init with Xiao, the cam is not activateable:
Cannot init camera
```
E (9509) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (9509) cam_hal: cam_config(385): cam_dma_config failed
E (9509) gdma: gdma_disconnect(299): no peripheral is connected to the channel
E (9516) camera: Camera config failed with error 0xffffffff
```

The sketch up to that stage is not that complex:
```
#include <dummy.h>

/**
 * Motion detection
 * Detect when the frame changes by a reasonable amount
 *
 * BE SURE TO SET "TOOLS > CORE DEBUG LEVEL = DEBUG"
 * to turn on debug messages
 */

 // WiFi credentials
#define WIFI_SSID "xxxx"
#define WIFI_PASS "yyyy"

#include <eloquent_esp32cam.h>
#include <eloquent_esp32cam/motion/detection.h>
#include <eloquent_esp32cam/extra/esp32/telegram.h>

using eloq::camera;
using eloq::motion::detection;
using eloq::wifi;
using eloq::telegram;

/**
 *
 */
void setup() {
    delay(3000);
    Serial.begin(115200);
    Serial.println("___MOTION DETECTION___");

    // camera settings
    // replace with your own model!
    camera.pinout.Xiao();
    camera.brownout.disable();
    camera.resolution.uxga();
    camera.quality.high();

    // configure motion detection
    // the higher the stride, the faster the detection
    // the higher the stride, the lesser granularity
    detection.stride(1);
    // the higher the threshold, the lesser sensitivity
    // (at pixel level)
    detection.threshold(8);
    // the higher the threshold, the lesser sensitivity
    // (at image level, from 0 to 1)
    detection.ratio(0.2);
    // optionally, you can enable rate limiting (aka debounce)
    // motion won't trigger more often than the specified frequency
    detection.rate.atMostOnceEvery(5).seconds();

    // init camera
    while (!camera.begin().isOk())
        Serial.println(camera.exception.toString());

    while (!wifi.connect().isOk())
      Serial.println(wifi.exception.toString());

    Serial.println("Camera OK");
    Serial.println("Awaiting for motion...");
}
```

I checked the Pins, which are correct from the docu for camera.pinout.Xiao();.

Eloquent ESP32cam Version 2.7.15
ESP32 Version 2.0.17

Not sure if this board needs something in addition maybe? I'm relative new to ESP32 ;)

BR
MP

eloquentarduino, 2025-02-18:
Are you able to run any other sketch from the library? Like the "Take_Picture.ino".

KaiyoFox, 2025-02-26:
E (9509) cam_hal: cam_dma_config(301): frame buffer malloc failed
Indicates PSRAM isn't setup right