eloquentarduino/EloquentEsp32cam #50: Repetitive Startup Sequence on Camera Initialization with camera.pinout.xiao()

KaiyoFox, 2025-02-24:
I am experiencing an issue with the XIAO ESP32-S3 where the startup sequence enters a repetitive loop during camera initialization. The example code that simply prints the camera's frame resolution fails to run correctly, and the device seems to reset continuously.

The Take_Picture.ino and Record_Video.ino examples (from another lib) work without any issues. However, when using the frame resolution example, the following log output is observed:

```ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376a24
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
```

I am using the `camera.pinout.xiao();` function for pin configuration, but the device enters a reset loop upon camera initialization.

I would appreciate any assistance in identifying the cause of this issue and any possible fixes or recommendations for resolving it.

KaiyoFox, 2025-02-24:
Yes, I enabled OPI PSRAM.

eloquentarduino, 2025-02-24:
Please paste here the sketch you're using.

KaiyoFox, 2025-02-24:
```cpp
#include <eloquent_esp32cam.h>

using eloq::camera;

void setup() {
    delay(3000);
    Serial.begin(115200);
    Serial.println("___GET YOUR FIRST PICTURE___");
    
    camera.pinout.xiao();
    camera.brownout.disable();
    camera.resolution.vga();
    
    while (!camera.begin().isOk())
        Serial.println(camera.exception.toString());
}

void loop() {
    if (!camera.capture().isOk()) {
        Serial.println(camera.exception.toString());
        return;
    }
    
    Serial.printf(
        "JPEG size in bytes: %d. Width: %dpx. Height: %dpx.\n",
        camera.getSizeInBytes(),
        camera.resolution.getWidth(),
        camera.resolution.getHeight()
    );
}
```

eloquentarduino, 2025-02-24:
Ok, looks like a basic sketch. Are you able to run any other sketch from my library? Also, please paste your config from the tools menu (psram, log level, CDC...)

KaiyoFox, 2025-02-24:
I haven't been able to run any other examples

[image]

eloquentarduino, 2025-02-24:
Try to set the board as ESP32S3 Dev Module, instead of xiao in the tools menu

KaiyoFox, 2025-02-24:
I unfortunately still have the same issue,
I tried OPI PSRAM and QSPI PSRAM, both had issues

KaiyoFox, 2025-02-25:
I may not have 2.x installed
Im on Arch Linux; and likely installed 3.x as a dependency when installing Arduino IDE

I can install ESP32 Core 2.x, But I don't think Arduino IDE will use it

eloquentarduino, 2025-02-25:
Usually when you install a new version, it overwrites the older (at least on OSX and Ubuntu)

KaiyoFox, 2025-02-25:
understood
and its not the Esp32 Board in Arduino IDE's Board Manager?

[image]

eloquentarduino, 2025-02-25:
Yes, use the drop-down to select the latest 2.* Version

KaiyoFox, 2025-02-26:
Thanks much, It work great now!

closed 2025-02-26 by KaiyoFox