eloquentarduino/EloquentEsp32cam #53: collect image for edge impulse is not working ( Due to old flush() code, may be required clear() in server.h & mpeg.h

ofsanjay1, 2025-04-07:
In file included from C:\Users\SanjayKumar\Documents\Arduino\libraries\EloquentEsp32cam\src/eloquent_esp32cam/viz/image_collection.h:7,
                 from C:\Users\SANJAY~1\AppData\Local\Temp\arduino_modified_sketch_413018\Collect_Images_for_EdgeImpulse.ino:22:
C:/Users/SanjayKumar/Documents/Arduino/libraries/EloquentEsp32cam/src/eloquent_esp32cam/extra/esp32/http/server.h: In member function 'void Eloquent::Extra::Esp32::Http::HttpServer::sendGzip(const uint8_t*, size_t)':
C:/Users/SanjayKumar/Documents/Arduino/libraries/EloquentEsp32cam/src/eloquent_esp32cam/extra/esp32/http/server.h:237:41: warning: 'virtual void NetworkClient::flush()' is deprecated: Use clear() instead. [-Wdeprecated-declarations]
  237 |                             client.flush();
      |                             ~~~~~~~~~~~~^~
In file included from C:\Users\SanjayKumar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\Network\src/Network.h:12,
                 from C:\Users\SanjayKumar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\WiFi\src/WiFiGeneric.h:44,
                 from C:\Users\SanjayKumar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\WiFi\src/WiFiSTA.h:30,
                 from C:\Users\SanjayKumar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\WiFi\src/WiFi.h:34,
                 from C:\Users\SanjayKumar\Documents\Arduino\libraries\EloquentEsp32cam\src/eloquent_esp32cam/extra/esp32/wifi/sta.h:4,
                 from C:\Users\SANJAY~1\AppData\Local\Temp\arduino_modified_sketch_413018\Collect_Images_for_EdgeImpulse.ino:21:
C:\Users\SanjayKumar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\Network\src/NetworkClient.h:61:8: note: declared here
   61 |   void flush();  // Print::flush tx
      |        ^~~~~
In file included from C:\Users\SanjayKumar\Documents\Arduino\libraries\EloquentEsp32cam\src/eloquent_esp32cam/viz/image_collection.h:8:
C:/Users/SanjayKumar/Documents/Arduino/libraries/EloquentEsp32cam/src/eloquent_esp32cam/viz/mjpeg.h: In lambda function:
C:/Users/SanjayKumar/Documents/Arduino/libraries/EloquentEsp32cam/src/eloquent_esp32cam/viz/mjpeg.h:155:45: warning: 'virtual void NetworkClient::flush()' is deprecated: Use clear() instead. [-Wdeprecated-declarations]
  155 |                                 client.flush();
      |                                 ~~~~~~~~~~~~^~
C:\Users\SanjayKumar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\Network\src/NetworkClient.h:61:8: note: declared here
   61 |   void flush();  // Print::flush tx
      |        ^~~~~
Sketch uses 1057496 bytes (33%) of program storage space. Maximum is 3145728 bytes.
Global variables use 53820 bytes (16%) of dynamic memory, leaving 273860 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.8.1
Serial port COM3
Connecting....
Chip is ESP32-D0WDQ6 (revision v1.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:4f:00:ad:25:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00112fff...
Compressed 25040 bytes to 16006...
Writing at 0x00001000... (100 %)
Wrote 25040 bytes (16006 compressed) at 0x00001000 in 0.8 seconds (effective 258.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 137...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (137 compressed) at 0x00008000 in 0.1 seconds (effective 467.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 633.9 kbit/s)...
Hash of data verified.
Compressed 1057600 bytes to 665774...
Writing at 0x00010000... (2 %)
Writing at 0x0001c045... (4 %)
Writing at 0x00028ea0... (7 %)
Writing at 0x0002f946... (9 %)
Writing at 0x0003fcc4... (12 %)
Writing at 0x00045c01... (14 %)
Writing at 0x0004bbe1... (17 %)
Writing at 0x000510c6... (19 %)
Writing at 0x000566d8... (21 %)
Writing at 0x0005bd74... (24 %)
Writing at 0x00060fa7... (26 %)
Writing at 0x000666a2... (29 %)
Writing at 0x0006bd9a... (31 %)
Writing at 0x000713e5... (34 %)
Writing at 0x00076e5f... (36 %)
Writing at 0x0007c210... (39 %)
Writing at 0x000818e7... (41 %)
Writing at 0x00086dc3... (43 %)
Writing at 0x0008d2f6... (46 %)
Writing at 0x0009376f... (48 %)
Writing at 0x00098c24... (51 %)
Writing at 0x0009e66b... (53 %)
Writing at 0x000a433b... (56 %)
Writing at 0x000a9b65... (58 %)
Writing at 0x000af370... (60 %)
Writing at 0x000b44e5... (63 %)
Writing at 0x000b9841... (65 %)
Writing at 0x000bef38... (68 %)
Writing at 0x000c4810... (70 %)
Writing at 0x000c9b17... (73 %)
Writing at 0x000cf0c4... (75 %)
Writing at 0x000d4ebe... (78 %)
Writing at 0x000daad4... (80 %)
Writing at 0x000e0d5e... (82 %)
Writing at 0x000eb510... (85 %)
Writing at 0x000f105d... (87 %)
Writing at 0x000f6134... (90 %)
Writing at 0x000fd75c... (92 %)
Writing at 0x00103104... (95 %)
Writing at 0x001089d3... (97 %)
Writing at 0x0010e4f3... (100 %)
Wrote 1057600 bytes (665774 compressed) at 0x00010000 in 16.7 seconds (effective 507.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

salernosimone, 2025-04-07:
This library is not compatible with ESP32 core 3.x. You have to downgrade to 2.x

salernosimone, 2025-04-08:
Be sure you enabled PSRAM, that's the most common issue. Also, beware that the common ESP32-CAM board is not recommended for this library, since it has low RAM and often reboots. It still should work, but gives a lot of troubles.

ofsanjay1, 2025-04-08:
Thank you for the response. Is i am able to run the ai model on it. i saw many videos that it works. will you have some guide website or article to go through to before working on it. i am little fond of adding most of the capability on small microcontroller. 
Again, thank you for the fast response.

salernosimone, 2025-04-08:
Follow this tutorial and be sure you can run all the examples on the page first: https://eloquentarduino.com/posts/esp32-cam-quickstart

If you get any errors, enable DEBUG logging (Tools menu > Log level > DEBUG) and report here.

ofsanjay1, 2025-04-08:
still having the same problem with the collect image for edge impulse code. The PSRAM is enabled but it is still not working. The code is being uploaded properly but not working as required like a server for collecting image and uploading it for local model running. thanks for the last response
[image]