eloquentarduino/EloquentEsp32cam #56: Error using Motion Detection

andiaa734, 2025-04-26:
I want to implement Motion Detection.

For a time the program runs fine. But then it throws an error message:

esp_jpg_decode: JPG Decompression Failed! Data format error

After this, the program hangs in the while loop:

if (!detection.run().isOk()) {
    Serial.println("Detection error: ");
    Serial.println(detection.exception.toString());
    return;
}

No more exception is printed. ESP needs no be reset to work again for a time.

Any solution? Tried to implement ESP.restart in the while loop but this conflicts with detection.rate.atMostOnceEvery(5).seconds();
So the ESP is restarted immediately after start.

andiaa734, 2025-04-27:
Thanks but unfortunately this is also not working. Still the same error. My code is now:

if (!camera.capture().isOk()) {
    Serial.println("Capture error: ");
    Serial.println(camera.exception.toString());
    ESP.restart();
    return;
}
 
if (camera.frame->buf[0] != 0xFF || camera.frame->buf[1] != 0xD8) {
  Serial.println("Frame doesn't look like a valid jpeg");
  return;
}
// run motion detection
if (!detection.run().isOk()) {
    Serial.println("Detection error: ");
    Serial.println(detection.exception.toString());
    camera.capture();
    return;
}

Serial.println("Frame doesn't look like a valid jpeg"); is not printed. It goes directly to esp_jpg_decode: JPG Decompression Failed! Data format error.

This only happens when using Motion detection. Running a stream with a webserver works without any interruption.

salernosimone, 2025-04-27:
It looks like the jpeg is corrupted later in the buffer. Let me think about it and I will come back whit an idea

andiaa734, 2025-04-27:
Thank you. i have now moved the main code which is called after motion is detected to a function. this makes it a little bit better. But still not running stable.

salernosimone, 2025-04-30:
I looked at the code for motion detection. The error is likely caused by the line

```c++
camera->mutex.threadsafe([this]() {
    if (!jpg2rgb565(camera->frame->buf, camera->frame->len, (uint8_t*) data, JPG_SCALE_8X))
        exception.set("Error converting frame from JPEG to RGB565");
});
```

in `src/eloquent_esp32cam/camera/rgb_565.h#L108`, but I don't see how the library would mess this up. `buf` and `len` are straight from the `camera_frame_t` struct that you get from the camera.

You can try to open that file and remove the `camera->mutex.threadsafe([this]() {}` wrapper, leaving only the `if` statement. I don't think it will make any difference, but that's the only thing I can think of for now.