eloquentarduino/EloquentEsp32cam #58: Support for the new ESP32S3 Sense Seed Studio Board

aditya-rao-iit-m, 2025-05-06:
Hi,
I created an addition for the pinout for the new Seed Studio board that we will be using in our college event.
Will submit a pull request.

eloquentarduino, 2025-06-16:
> When I use Seeed Studio XIAO ESP32-S3 Sense,
>
> Collect_Images_for_EdgeImpulse，I edit the code
>
>     // camera settings
>     // replace with your own model!
>     camera.pinout.xiao();
>     camera.brownout.disable();
>
> E (47493) cam_hal: cam_config(384): cam_dma_config failed
> E (47493) camera: Camera config failed with error 0xffffffff
> Cannot init camera
> E (47591) cam_hal: cam_dma_config(300): frame buffer malloc failed
> E (47592) cam_hal: cam_config(384): cam_dma_config failed
> E (47592) camera: Camera config failed with error 0xffffffff
> Cannot init camera
> E (47690) cam_hal: cam_dma_config(300): frame buffer malloc failed
> E (47691) cam_hal: cam_config(384): cam_dma_config failed
> E (47691) camera: Camera config failed with error 0xffffffff
> Cannot init camera
> E (47789) cam_hal: cam_dma_config(300): frame buffer malloc failed
> E (47790) cam_hal: cam_config(384): cam_dma_config failed
> E (47790) camera: Camera config failed with error 0xffffffff
> Cannot init camera
> E (47888) cam_hal: cam_dma_config(300): frame buffer malloc failed
> E (47889) cam_hal: cam_config(384): cam_dma_config failed
> E (47889) camera: Camera config failed with error 0xffffffff
> Cannot init camera
> E (47987) cam_hal: cam_dma_config(300): frame buffer malloc failed
> E (47988) cam_hal: cam_config(384): cam_dma_config failed
> E (47988) camera: Camera config failed with error 0xffffffff
> Cannot init camera
> E (48086) cam_hal: cam_dma_config(300): frame buffer malloc failed
> E (48087) cam_hal: cam_config(384): cam_dma_config failed
> E (48087) camera: Camera config failed with error 0xffffffff
>
> I got the logs above.

Be sure to select ESP32S3 Dev Module as board model, not XIAO ESP32S3 from
the Board menu.

Also, try to delete the brownout line

typetrade, 2025-06-18:
> Did you attached the external antenna to the board? Also, double check the
> SSID is written correctly.
> 
> Il mar 17 giu 2025, 03:40 steven ***@***.***> ha scritto:
> […](#)

The SSID is correct, this is my phone hotspot.
I modified the configuration of Arduino and changed the PSRAM Disable in the tools menu to OPI PSRAM, which solved my problem.